BCN3D/BCN3DSigma-Profile-Generator #2: Bed / Hot End Heating Sequence Should be Sequential, not Parallel

dpersuhn, 2017-01-25:
The profile generator starts the heating process for the bed and hot ends simultaneously, resulting in the hot ends arriving at the target temperature long before the bed.  In some cases, keeping a hot end at extrusion temperature without actually extruding material can result in a blocked nozzle due to the filament losing its flow properties.  The heating sequence should bring the bed to temperature before starting the hot end.  If the goal is to minimize heating time, perhaps the hot end could start heating when the bed is within a few degrees of its target temperature?

guillemavila, 2017-02-01:
That's right, I'll update the sequence for high bed temperatures.
For lower ones bed and hotends reach the target almost in sync so it's faster to work in parallel.

guillemavila, 2017-02-14:
Just updated the start sequence with the new version.

I went a bit further and made the algorithm to study target temperatures and define a custom sequence for every print. It starts heating each element at the right moment so everything reach it's own target in sync.

closed 2017-02-16 by guillemavila