BCN3D/BCN3DSigma-Profile-Generator #7: Extrusion width smaller than nozzle diameter ???

mrcg, 2017-10-05:
In latest (05.10.2017) generated profile the Extrusion Width is smaller than the nozzle diameter.
E.g. 
Nozzle Diameter: 0,4mm
Extrusion Width: 0,35mm

Honestly speaking this does not make any sense.
Extrusion width must be minimum nozzle diameter or larger.

Especially for prints with more shells (e.g. >3 shells) this will cause issues.
For me this is a bug and needs to be corrected. This value should also not used for horizontal size compensation.

guillemavila, 2017-11-09:
Turned back to nozzle diameter, thanks

closed 2017-11-09 by guillemavila

thesec, 2017-11-10:
Actually there are quite a few filaments extruding smaller than the nozzle diameter with non-modified extrusion settings.
If BCN3D is aware of these and this is reflected in the profiles that is much appreciated.

The other way around - if you would like settings to get these filaments to extrude in nozzle width - has a lot more dependencies and efforts to it and is likely to fail with the first setting changed (by the unexperienced user).

Imho not a bug but a feature where the filament does indeed extrude with a smaller diameter.

commit 3555bcc6 (2017-11-09): Closes #7