BCN3D/BCN3DSigma-Profile-Generator #8: ProGen v2.0.2 profiles no longer work in S3D 4.0

hst68, 2017-11-01:
The S3D ProGen Profiles seem to me to be a bit flawed. See screenshots.
I put together the following compilation on ProGen website:
Left Extruder 0.3
Right Extruder 0.5
In S3D I only get the Left Extruder.

other configurations, e. g. only right-hand extruders, do not work. Also the Nozzles diameters are not copied correctly.
The ProGens from last week are all working.

[image]
[image]

guillemavila, 2017-11-03:
Hello, all profiles are working for us. Did you try to remove the old imported ones and import the new ones again?

thesec, 2017-11-04:
@hst68 S3D is flawed itself. Use a working and actively supported slicer.

hst68, 2017-11-04:
The problem was that the profiles of S3D were not deleted correctly from the database. 
They weren't displayed anymore. But with the new add, all profiles didn't work anymore. Even if you uninstall S3D, the. plist will remain in the library.
Only when I have deleted the following entries via terminal. I was able to install S3D cleanly again and read the FFF profiles correctly!

For completeness if you still want to do a complete uninstall:
[*] Run the built in uninstall and choose to remove the license. I like to play it safe and not have a license tied up on a non-operational machine. Although there is an option to reclaim the license online for the current version should you forget.
[*] Run the following 2 commands in terminal.
[*] rm -rf /Users/$USER/Library/Application Support/Simplify3D
[*] rm /Users/$USER/Library/Preferences/com.simplify3d.S3D-Software.plist

guillemavila, 2017-11-05:
Thanks for sharing, I've reported the issue to S3D. Hope they fix it soon :)

closed 2017-11-05 by guillemavila