BCN3D/BCN3DSigma-Profile-Generator #10: "print_mode" is unknown

thesec, 2018-03-11:
"print_mode" is unknown to Ultimaker Cura, the generated printer definition fails loading with it.

Please remove these and add an Ultimaker Cura profile to generate for.

guillemavila, 2018-03-14:
Sad to say but this is not in our roadmap. This tool is a Profile Generator for BCN3D Cura & Simplify3D slicing softwares. As the readme states.

closed 2018-03-14 by guillemavila

thesec, 2018-03-14:
Sad that you do not even read your own readme: "Profile Generator for Simplify3D, Cura & Cura 2 slicing softwares".

guillemavila, 2018-03-15:
*The readme of the Development branch*

From now on, I want to tell you we are all here for the same reason: make things better and be happier as a byproduct of that. We share almost everything we do. If you are not satisfied with anything it's cool to share it, but if you expect some action from us [maybe you should change a bit the way you write](https://www.amazon.com/How-Win-Friends-Influence-People/dp/0671027034).

Thanks for understanding. Peace

thesec, 2018-03-15:
[Peace](https://www.amazon.com/Power-Positive-Thinking-Norman-Vincent/dp/0743234804/) into the growing walled garden.